thespacedoctor/soxspipe #14: Pipeline document

marcolandoni, 2019-10-17:
Prepare a TeX file for each receipe starting from the previous meeting.

thespacedoctor, 2019-10-17:
Purpose

Input Table - include where frame originated from

Recipe Logic Steps

Output Table

Later ... workflow diagrams, image examples ...

thespacedoctor, 2019-10-17:
Workflow Procedures:

Purpose

Input Table - include where frame originated from

Procedures Logic Steps

Output Table

thespacedoctor, 2019-10-17:
Initial Tasks:

- [ ] Prepare template - DY
- [ ] Transefer text from old doc

Recipes

- [ ] soxs_data_organise - DY
- [ ] soxs_mbias - DY
- [ ] soxs_disp - ML
- [ ] soxs_order_pos - DY
- [ ] soxs_spec_mflat - DY
- [ ] soxs_wave - ML
- [ ] soxs_2d_rectify - ML

Procedure

- [ ] p_soxs_prepare - DY
- [ ] p_subtract_backgroud - ML
- [ ] p_order_edges - ML

Other tasks:

- [ ] update cascade diagrams
- [ ] add recipes for A&G cam
- [ ] add workflow diagram for each recipe

marcolandoni, 2020-01-31:
Put into the presented document at ESO on Nov 2019 the logic (very shallow)  of the "black pages recipe".

closed 2021-05-20 by thespacedoctor