thespacedoctor/soxspipe #34: soxs order centre - write the logic to juptyer notebook

thespacedoctor, 2020-01-30:
Starting from the X and Y polynomials for each order resulting from `soxs-predict`, write code to capture the central order positions from the single-pinhole lamp flat frames.

This may be from generating an array of pixels in say the Y directory (e.g. 1-2048) and for each column measure the X-pixel where a 1D gaussian best fits the profile of the continuum.

marcolandoni, 2020-03-06:
Logic under development.
Following strictly xsh_follow_continuum

marcolandoni, 2020-08-20:
Logic completed. 
Defining QCs and residuals, moving the new soxs_predict interface to soxs_ordercentre

marcolandoni, 2020-09-04:
Logic completed. Jupyter notebook updated.
Need to define QCs, but in a later stage.

closed 2020-09-04 by marcolandoni