thespacedoctor/soxspipe #73: Bias QC

marcolandoni, 2021-08-23:
**Here below the QCs for the detector:bias group**

- Master BIAS median. The bias level (in ADU) is evaluated as the median value of all pixels in the master bias.

- RON-raw: The difference between the first and the second input raw biases (in ADU) is computed. The population standard deviation of the pixel values of this difference, divided by the square root of 2, is taken as the RON.

marcolandoni, 2021-08-23:
- RON master: For the master an expected master bias noise is calculated from the ron of the raw data, divided by the square root of the number of frames that were used to create the master bias. Then the population standard deviation of the pixel values of the master bias is determined, excluding all pixels that differ more than three times the expected ron from the median level of the master bias.

- Struct X: To determine the structure along the X/Y-axis the bias is collapsed along the Y/X-axis and the slope of the resulting one-dimensional frame is determined. 

- Struct Y: To determine the structure along the X/Y-axis the bias is collapsed along the Y/X-axis and the slope of the resulting one-dimensional frame is determined. 

- Per-Pat-Noise: Since 2021 the xshooter_mbias recipe is executed with the --pd_noise_compute=TRUE command line option that performs a 2D Fourier transform of each of the five raw BIAS frames and calculates the (not robust) standard deviation and (robust) MAD (median absolute deviation). Both parameters are ingested into the QC1 DB, and the maximum of the five std/stdMAD values (qc_PN_maxrat in the DB) is monitored here .

thespacedoctor, 2021-08-31:
[UVB XShooter Page ](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_BIAS_UVB_med_master_HC.html
)

[VIS XShooter Page ](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_BIAS_VIS_med_master_HC.html)

thespacedoctor, 2021-12-01:
Notebook here: https://soxs-eso-data.org/user/dry/lab/workspaces/auto-V/tree/shared-notebooks/QCs/bias_qcs/bias_qcs.ipynb

marcolandoni, 2022-01-05:
Implemented correctly. Values are comparable (within order of magnitudes) with the historical value reported in the XSHOOTER Health check monitor

thespacedoctor, 2022-01-13:
Next task is to integrate these QCs into the main codebase. We can do this together.

thespacedoctor, 2022-03-03:
@thespacedoctor to integrate Per-Pat-Noise into main code

thespacedoctor, 2022-03-10:
Here is an example of the 2D FFT of a single raw bias frame (using absolute values to remove the complex components):

[image]

The maximum std/mad is now calculated and stored as QC. 

[image]

thespacedoctor, 2022-03-11:
Sigma-clipping the 2D FFT image before calculating std and mad removes a single extreme value (centre pixel?) that highly skews the std, but not mad value. Calculated ratios are now much closer to those measured in the XShooter health check monitor.

commit f4dc2fbe (2022-03-11): added period pattern noise QC calc to mbias

closes #73
closed 2022-03-11 by thespacedoctor