thespacedoctor/soxspipe #75: Distortion Correction QCs

marcolandoni, 2021-08-23:
- Number identified: 

1. the number of identified arc lines in (RAW_TYPE = FMTCHK, recipe = xsh_predict, 1 single-pinhole) frames
2. the number of identified arc lines in (RAW_TYPE = WAVE, recipe = xsh_2dmap, 1 9-pinhole arc) frames
3. the number of identified orders (RAW_TYPE = ORDERDEF, recipe = xsh_orderpos, 1 flat and 1 single-pinhole) frames

- Gaps
Largest gap (in pixels) between identified lines in FMTCHK (1-pinhole arc) and 2DMAP (9-pinhole arc) data. We also show the number of the order in which the gap appears

- fmt_resid: Residuals of the lines in FMTCHK (1-pinhole arc) data after the annealing of the physical model. residx/y_min/max may be large due to single outliers and are therefore not scored.

-  Residuals: residuals of the ORDERDEF (1-pinhole flat) data after the fitting (min, max, rms)

- 2d residual: Residuals of the lines in 2DMAP (9-pinhole arc) data after the annealing of the physical model. residx/y_min/max may be large due to single outliers and are therefore not scored.

thespacedoctor, 2021-08-31:
[UVB XShooter Page](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_NLINE_UVB_HC.html)

[VIS XShooter Page](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_NLINE_VIS_HC.html)

[NIR XShooter Page](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_NLINE_NIR_HC.html)

marcolandoni, 2021-09-09:
Ask Dave data needed to be retrieved in the fits header.

thespacedoctor, 2021-09-12:
Hi @marcolandoni, I think we may waste effort re-coding this in a notebook. I'm calculating 70% of these QCs already in the main code-base. I suggest we use this Thursday's meeting to a) get the existing QC values written neatly into the product FITS headers (with an agreed naming format) and b) code the missing QCs (if we have time). All straight into branch of the code.

This can be out first test case to make sure the QC container (persistent pandas data frame) is working.

marcolandoni, 2022-01-05:
I agree! Let's do it with the integration of Bias QC in the next iteration togheter!

thespacedoctor, 2022-04-12:
I've now added QCs to FMTCHK (1-pinhole arc) data. Dropping the 'Gaps' QCs as I'm not sure how useful they are. But I have added combined XY residuals alongside the required single-axis residuals. I have also added a 'PLINE' QC that gives the proportion of lines detected (compared to static line list table) as I think this is more informative than just stating the number of lines detected.

[image]

closed 2022-04-13 by thespacedoctor