thespacedoctor/soxspipe #79: Efficiency QCs

marcolandoni, 2021-08-23:
**Here below the relevant QCs for the efficiency group**

- efficiency: Relative efficiency for UVB arm, range 5100A - 5300A, EFFMON template, started under photometric conditions, Chromatic efficiency plot for UVB arm; the values trended in plot are marked. The plot is constituted from the following wavelength ranges: 5670A, a point each circa 200 angstroms (chromatic efficiency)

-flat_response: For each calibration flat lamp: D2 current response / reference response = a + b * (lam - lam0), D2 current response / reference response = a + b * (lam - lam0), 

> On 2015-10-25, the first flux standard star was acquired with the effmon template, meaning this observation was acquired under strict photometric conditions. The calibration product 'response function' from that date serves as a reference in the following. It is marked by a vertical red bar in the history plot.
> 
> The master response function once generated as a weighted and clean mean of response functions for the purpose to flux calibrate XSHOOTER science observations (internal data producs = IDPs) is a representative response function for a given time range. The comparison between the reference response function (from 2015-10-25, ARCFILE=XSHOO.2015-10-26T01:23...) and the current response function as well as the comparison between the master response and the current response is parameterized via
> 
> 
> 
> current(lambda) / reference(lambda) = a + b * (lambda - lambda_0 )
> 
> current(lambda) / master(lambda) = a + b * ( lambda - lambda_0 )
> 
> where lambda_0 is the center wavelength of the considered wavelength range, parameter a is the relative overall flux of the spectrum, corresponding to atmospheric variations and parameter b is a first order chromatic deviation (the slope between the reference(/master) and the current).
> 
> This plot shows both parameters a and b, of which parameter b is the important one.
> 
> On 2015-10-25, there is a=1 and b=0 by definition. The response function is a flat-fielded efficiency, hence the response function depends on the spectral shape of the flat field lamp, on long-term changes as well as on exchanges of the lamp. For this reason, dates of lamp exchanges are marked by blue vertical bars.

thespacedoctor, 2021-08-31:
[UVB XShooter Page](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_EFF_UVB_HC.html)

[VIS XShooter Page](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_EFF_VIS_HC.html)

[NIR XShooter Page](https://www.eso.org/observing/dfo/quality/XSHOOTER/reports/HEALTH/trend_report_EFF_NIR_HC.html)

thespacedoctor, 2021-09-12:
FYI I added '3. median star counts in arm at given range' to the list above