thespacedoctor/soxspipe #114: Respond to AM report on pipeline

thespacedoctor, 2022-05-03:


smarttgit, 2022-05-06:
Some comments and points I took from the SOXS progress review meeting (5th May 2022), and two points from me as I was reviewing the demo software. 

soxspipe does not currently use the value of the bias in the overscan region. It subtracts the master bias only. Most pipelines will check the value in the overscan. An simple offset can be applied if the bias level has drifted from the value in the masterbias. Would be good to implement this. It's fairly standard for CCD reductions 

QC plots for commissioning - FWHM plots for the 2D frame will be very   useful to check focus and line-spread function across the two chips. 

Cosmic rays in the CCD - will we be using a version of LaCosmic (Astroscrappy) to clean the chip of cosmic rays before the extraction ? If we do, this is often time intensive, so we only need to clean the UV-VIS CCD in the central region around the orders (which is what the PESSTO pipeline does), not the full chip. Let's check the NIR detector. 

Wavelength calibration - identification of lines and solution looks like it's working well. We'll need to consider how we can tweak a solution to map an arc solution taken early in the night to all the exposures through identification of skylines (emission - for long exposures; or telluric absorption for short exposures of bright sources). 

Just making a note here, and we can discuss and elevate these to individual issues

thespacedoctor, 2022-05-13:
Doc to collect responses here: https://docs.google.com/document/d/16hmx3dSwPnbRn97C0unGg8rivlf8eR7OOvI7-tmaEw4/edit?usp=sharing

closed 2025-02-26 by thespacedoctor