thespacedoctor/soxspipe #154: Pipeline crashing while computing flats

jbodenst, 2022-11-08:
I just installed the new version of the pipeline and downloaded the demo data. While running the "run_pipeline.sh" script, it shows quite a lot of warnings and crashes at some point during the flats .. 

Attached are the outputs of the pipeline:

> * 10:19:45 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py:main:94 > --- STARTING TO RUN THE cl_utils.py AT 2022-11-08T10:19:45
> * 10:19:46 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:47 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:48 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:48 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:49 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> # VERIFYING INPUT FRAMES
> [1A[2K# VERIFYING INPUT FRAMES - ALL GOOD
> 
> # PREPARING 5 RAW FRAMES - TRIMMING OVERSCAN, CONVERTING TO ELECTRON COUNTS, GENERATING UNCERTAINTY MAPS AND APPENDING DEFAULT BAD-PIXEL MASK
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> # PREPARED FRAMES - SUMMARY
>                filename                INSTRUME ... ESO SEQ ARM EXPTIME
> -------------------------------------- -------- ... ----------- -------
> XSHOO.2019-08-30T11:23:13.765_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-30T11:26:03.489_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-30T11:28:53.494_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-30T11:31:43.499_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-30T11:34:33.504_pre.fits XSHOOTER ...         UVB     0.0
> Traceback (most recent call last):
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/bin/soxspipe", line 10, in <module>
>     sys.exit(main())
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py", line 157, in main
>     mbiasFrame = recipe.produce_product()
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_mbias.py", line 157, in produce_product
>     meanBiasLevels, rons, noiseFrames = zip(*[self.subtact_mean_flux_level(c) for c in ccds])
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_mbias.py", line 157, in <listcomp>
>     meanBiasLevels, rons, noiseFrames = zip(*[self.subtact_mean_flux_level(c) for c in ccds])
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/_base_recipe_.py", line 1203, in subtact_mean_flux_level
>     clipping_lower_sigma = self.settings[
> KeyError: 'clipping-lower-sigma'
> * 10:19:50 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py:main:94 > --- STARTING TO RUN THE cl_utils.py AT 2022-11-08T10:19:50
> * 10:19:51 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:52 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:53 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:54 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> * 10:19:55 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/astropy/nddata/ccddata.py:uncertainty:263 > array provided for uncertainty; assuming it is a StdDevUncertainty.
> # VERIFYING INPUT FRAMES
> [1A[2K# VERIFYING INPUT FRAMES - ALL GOOD
> 
> # PREPARING 5 RAW FRAMES - TRIMMING OVERSCAN, CONVERTING TO ELECTRON COUNTS, GENERATING UNCERTAINTY MAPS AND APPENDING DEFAULT BAD-PIXEL MASK
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> INFO: array provided for uncertainty; assuming it is a StdDevUncertainty. [astropy.nddata.ccddata]
> # PREPARED FRAMES - SUMMARY
>                filename                INSTRUME ... ESO SEQ ARM EXPTIME
> -------------------------------------- -------- ... ----------- -------
> XSHOO.2019-08-31T10:25:14.650_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-31T10:28:04.834_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-31T10:30:54.669_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-31T10:33:44.673_pre.fits XSHOOTER ...         UVB     0.0
> XSHOO.2019-08-31T10:36:32.538_pre.fits XSHOOTER ...         UVB     0.0
> Traceback (most recent call last):
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/bin/soxspipe", line 10, in <module>
>     sys.exit(main())
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py", line 157, in main
>     mbiasFrame = recipe.produce_product()
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_mbias.py", line 157, in produce_product
>     meanBiasLevels, rons, noiseFrames = zip(*[self.subtact_mean_flux_level(c) for c in ccds])
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_mbias.py", line 157, in <listcomp>
[... 480 lines cut ...]
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/commonutils/set_of_files.py", line 359, in get
>     raise FileNotFoundError(f"the input file `{f}` does not appear to exist")
> FileNotFoundError: the input file `./product/soxs-mbias/2019.08.31T10.25.14.650_UVB_1X1_SLOW_MBIAS.fits` does not appear to exist
> * 10:21:08 - INFO: /scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py:main:94 > --- STARTING TO RUN THE cl_utils.py AT 2022-11-08T10:21:08
> Traceback (most recent call last):
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/bin/soxspipe", line 10, in <module>
>     sys.exit(main())
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py", line 213, in main
>     recipe = soxs_stare(
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_stare.py", line 80, in __init__
>     self.inputFrames, self.supplementaryInput = sof.get()
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/commonutils/set_of_files.py", line 359, in get
>     raise FileNotFoundError(f"the input file `{f}` does not appear to exist")
> FileNotFoundError: the input file `./product/soxs-mbias/2019.08.31T10.26.28.446_VIS_1X1_SLOW_MBIAS.fits` does not appear to exist
[... cut ...]
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_stare.py", line 80, in __init__
>     self.inputFrames, self.supplementaryInput = sof.get()
>   File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/commonutils/set_of_files.py", line 359, in get
>     raise FileNotFoundError(f"the input file `{f}` does not appear to exist")
> FileNotFoundError: the input file `./product/soxs-spatial-solution/2019.08.30T18.43.48.7597_NIR_SPAT_SOL_0PT6651.fits` does not appear to exist

thespacedoctor, 2022-11-08:
Thanks @jbodenst. Could you please send me your `~/.config/soxspipe/soxspipe.yaml` file ? You can drag and drop the file into a comment field to share it.

jbodenst, 2022-11-08:
I had to rename it for github to allow me to drag it in here, but I hope this works :) 

[soxspipe.txt](https://github.com/thespacedoctor/soxspipe/files/9962095/soxspipe.txt)

jbodenst, 2022-11-08:
actually ... looking at the file myself I might see some possible causes for problems .. let me see if I can fix it

jbodenst, 2022-11-08:
I thought the path definitions in the .yaml file might have caused the issue (_intermediate-data-root_ and _reduced-data-root_) but that does not seem to solve the problem

thespacedoctor, 2022-11-08:
Ah yes, this is an old settings file. Please delete this file (`~/.config/soxspipe/soxspipe.yaml`) and run `soxspipe init` once more to get the latest settings.

Once you have the new settings file could you open it and replace the `soxs-mflat` block with:

```yaml
soxs-mflat:
    centre-order-window: 20
    stacked-clipping-sigma: 5
    stacked-clipping-iterations: 5
    clipping-lower-sigma: 50
    clipping-upper-sigma: 1000   
    clipping-iteration-count: 1
    slice-length-for-edge-detection: 90
    slice-width-for-edge-detection: 5
    min-percentage-threshold-for-edge-detection: 20
    max-percentage-threshold-for-edge-detection: 50
    disp-axis-deg: 3
    order-deg: 5
    poly-fitting-residual-clipping-sigma: 4
    poly-clipping-iteration-limit: 5
    low-sensitivity-clipping-simga: 2
```

I realised I had the default `poly-fitting-residual-clipping-sigma` too low.

After this please try and run the pipeline again :)

jbodenst, 2022-11-08:
Great, thanks :-) it's running ...

thespacedoctor, 2022-11-08:
Note that some speed improvements in image combination are due soon. The improvements are sitting in a development branch of a dependency codebase (ccdproc) and are due to be pushed into production shortly.

jbodenst, 2022-11-09:
unfortunately, the pipeline did not run through. This already happened with the last pipeline version, always at the step 
`# CREATING 2D IMAGE MAP FROM DISPERSION SOLUTION`
the pipeline crashes without any error message or warning, and the terminal I was running it from just disappears. This seems to me like a memory issue, but there is no error message like a "core dump" or something like this.

What are the minimum system requirements a laptop has to fulfill to run the pipeline?

thespacedoctor, 2022-11-09:
This step is absolutely the most memory and CPU intensive. We need to nail down the minimum requirements. Can I ask what spec of machine are you running the pipeline on?

jbodenst, 2022-11-09:
I have ... 8 cores (specifically: 11th Gen Intel® Core™ i7-1165G7 @ 2.80GHz × 8), 16 GB RAM, 8 GB Swp, and a 64-bit Fedora Linux 35 (Workstation Edition)

jbodenst, 2022-11-09:
Another thought .. could it have to do something with the "ulimit" settings?

thespacedoctor, 2022-11-10:
That's a good spec machine, similar to or better than other machines we test the pipeline on. So you might be onto something with the user limits. Can you send me the report given by `ulimit -a` please?

jbodenst, 2022-11-10:
yes sure, this is what I have:

```
(soxspipe_n) jbodenst@hql025887:~/material/SoXS/test_new_pipeline/soxspipe-quickstart-demo ulimit -a
real-time non-blocking time  (microseconds, -R) unlimited
core file size              (blocks, -c) unlimited
data seg size               (kbytes, -d) unlimited
scheduling priority                 (-e) 0
file size                   (blocks, -f) unlimited
pending signals                     (-i) 62366
max locked memory           (kbytes, -l) 8192
max memory size             (kbytes, -m) unlimited
open files                          (-n) 1024
pipe size                (512 bytes, -p) 8
POSIX message queues         (bytes, -q) 819200
real-time priority                  (-r) 0
stack size                  (kbytes, -s) 8192
cpu time                   (seconds, -t) unlimited
max user processes                  (-u) 62366
virtual memory              (kbytes, -v) unlimited
file locks                          (-x) unlimited

```

thespacedoctor, 2022-11-10:
The max locked memory size seems surprisingly low, on macOS it is unlimited by default. Can you please try and open this to ~1GB and test the pipeline again.

`ulimit -l 1000000`

thespacedoctor, 2022-11-10:
I think I have finally zone in on the issue. OpenBLAS (used by numpy) does not play well with multiprocessing and needs to be told to only run in single-thread mode. Numpy is compiled against MKL on my mac and not OpenBLAS and so I wasn't seeing the problem.

Before I update the code can you do a couple of tests for me please? Activate the python interpreter with:

```bash
conda activate soxspipe
python
```
 and then

```bash
import numpy
numpy.__config__.show()
```

Please paste the output here. Thanks!

jbodenst, 2022-11-11:
Hey David, here comes the output:

```
(std3) jbodenst@hql025887:~/material/SoXS/test_new_pipeline conda activate soxspipe_n 
(soxspipe_n) jbodenst@hql025887:~/material/SoXS/test_new_pipeline python
Python 3.8.13 | packaged by conda-forge | (default, Mar 25 2022, 06:04:18) 
[GCC 10.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy
>>> numpy.__config__.show()
blas_info:
    libraries = ['cblas', 'blas', 'cblas', 'blas']
    library_dirs = ['/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib']
    include_dirs = ['/scratch/home/jbodenst/.conda/envs/soxspipe_n/include']
    language = c
    define_macros = [('HAVE_CBLAS', None)]
blas_opt_info:
    define_macros = [('NO_ATLAS_INFO', 1), ('HAVE_CBLAS', None)]
    libraries = ['cblas', 'blas', 'cblas', 'blas']
    library_dirs = ['/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib']
    include_dirs = ['/scratch/home/jbodenst/.conda/envs/soxspipe_n/include']
    language = c
lapack_info:
    libraries = ['lapack', 'blas', 'lapack', 'blas']
    library_dirs = ['/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib']
    language = f77
lapack_opt_info:
    libraries = ['lapack', 'blas', 'lapack', 'blas', 'cblas', 'blas', 'cblas', 'blas']
    library_dirs = ['/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib']
    language = c
    define_macros = [('NO_ATLAS_INFO', 1), ('HAVE_CBLAS', None)]
    include_dirs = ['/scratch/home/jbodenst/.conda/envs/soxspipe_n/include']
Supported SIMD extensions in this NumPy install:
    baseline = SSE,SSE2,SSE3
    found = SSSE3,SSE41,POPCNT,SSE42,AVX,F16C,FMA3,AVX2,AVX512F,AVX512CD,AVX512_SKX,AVX512_CLX,AVX512_CNL,AVX512_ICL
    not found = AVX512_KNL,AVX512_KNM
```

thespacedoctor, 2022-11-11:
Looks like your numpy is compiled against BLAS. Next step is to place BLAS into single-threaded mode so it plays nicely with multiprocessing. Please try:

```bash
export OPENBLAS_NUM_THREADS=1
export BLAS_NUM_THREADS=1
soxspipe -x spat_sol ./sof/2019.08.22T14.16.36.0535_NIR_SPAT_SOL_0PT6651.sof -o ./
```

jbodenst, 2022-11-14:
Hey David, I set those options, but now I get a FileNotFoundError (see below) .. does that mean that the steps beforehand did not work out properly? Or is it just a filename I should change? 

`The product of this recipe already exists at './product/soxs-spatial-solution/2019.08.30T18.43.48.7597_NIR_SPAT_SOL_0PT6651.fits'. To overwrite this product, rerun the pipeline command with the overwrite flag (-x).
Traceback (most recent call last):
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/bin/soxspipe", line 10, in <module>
    sys.exit(main())
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py", line 213, in main
    recipe = soxs_stare(
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_stare.py", line 80, in __init__
    self.inputFrames, self.supplementaryInput = sof.get()
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/commonutils/set_of_files.py", line 420, in get
    raise TypeError(
TypeError: 'inputFrames' should be the path to a directory of files, an SOF file or a list of FITS frame paths
Traceback (most recent call last):
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/bin/soxspipe", line 10, in <module>
    sys.exit(main())
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py", line 213, in main
    recipe = soxs_stare(
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_stare.py", line 80, in __init__
    self.inputFrames, self.supplementaryInput = sof.get()
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/commonutils/set_of_files.py", line 359, in get
    raise FileNotFoundError(f"the input file `{f}` does not appear to exist")
FileNotFoundError: the input file `./product/soxs-spatial-solution/2019.08.30T18.42.31.199_VIS_1X1_FAST_SPAT_SOL_IMAGE.fits` does not appear to exist
Traceback (most recent call last):
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/bin/soxspipe", line 10, in <module>
    sys.exit(main())
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/cl_utils.py", line 213, in main
    recipe = soxs_stare(
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/recipes/soxs_stare.py", line 80, in __init__
    self.inputFrames, self.supplementaryInput = sof.get()
  File "/scratch/home/jbodenst/.conda/envs/soxspipe_n/lib/python3.8/site-packages/soxspipe/commonutils/set_of_files.py", line 359, in get
    raise FileNotFoundError(f"the input file `{f}` does not appear to exist")
FileNotFoundError: the input file `./product/soxs-spatial-solution/2019.08.30T18.43.48.7597_NIR_SPAT_SOL_0PT6651_IMAGE.fits` does not appear to exist
`

thespacedoctor, 2022-11-14:
I think you probably just need to run the previous 3 recipes with the `-x` flag to overwrite the products:

```bash
soxspipe -x spat_sol ./sof/2019.08.30T18.42.26.108_UVB_1X1_FAST_SPAT_SOL.sof -o ./
soxspipe -x spat_sol ./sof/2019.08.30T18.42.31.199_VIS_1X1_FAST_SPAT_SOL.sof -o ./
soxspipe -x spat_sol ./sof/2019.08.30T18.43.48.7597_NIR_SPAT_SOL_0PT6651.sof -o ./
```

If this doesn't work we can jump on a call and get things going again :)