thespacedoctor/soxspipe #170: soxspipe v0.8.0 installation issue

smarttgit, 2023-05-12:
```
(soxspipe) bash$/Users/sjs/Dropbox/data>soxspipe -v
v0.6.2
(soxspipe) bash$/Users/sjs/Dropbox/data>conda upgrade soxspipe -c coda-forge
(soxspipe) bash$/Users/sjs/Dropbox/data>conda upgrade soxspipe -c conda-forge
Collecting package metadata (current_repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.11.0
  latest version: 23.3.1

Please update conda by running

    $ conda update -n base -c defaults conda

## Package Plan ##

  environment location: /Users/sjs/opt/anaconda3/envs/soxspipe

  added / updated specs:
    - soxspipe

The following packages will be downloaded:
```
.... then went through all fine .... but check fails : 

```
(soxspipe) bash$/Users/sjs/Dropbox/data>soxspipe -v
Traceback (most recent call last):
  File "/Users/sjs/opt/anaconda3/envs/soxspipe/bin/soxspipe", line 6, in <module>
    from soxspipe.cl_utils import main
  File "/Users/sjs/opt/anaconda3/envs/soxspipe/lib/python3.8/site-packages/soxspipe/__init__.py", line 6, in <module>
    from . import recipes
  File "/Users/sjs/opt/anaconda3/envs/soxspipe/lib/python3.8/site-packages/soxspipe/recipes/__init__.py", line 5, in <module>
    from .soxs_mbias import soxs_mbias
  File "/Users/sjs/opt/anaconda3/envs/soxspipe/lib/python3.8/site-packages/soxspipe/recipes/soxs_mbias.py", line 29, in <module>
    from scipy.stats import median_absolute_deviation
ImportError: cannot import name 'median_absolute_deviation' from 'scipy.stats' (/Users/sjs/opt/anaconda3/envs/soxspipe/lib/python3.8/site-packages/scipy/stats/__init__.py)
```

```
(soxspipe) bash$/Users/sjs/Dropbox/data>python3 -V
Python 3.8.16
(soxspipe) bash$/Users/sjs/Dropbox/data>conda -V
conda 4.11.0
```
I guess I need to upgrade both ?

thespacedoctor, 2023-05-12:
The root issue is likely conda & anaconda are very out of date.

```bash
conda deactivate
conda update conda
conda update anaconda
```

Then can you try and install soxspipe from scratch:

```bash
rm ~/.config/soxspipe/soxspipe.yaml
conda deactivate
conda create -n soxspipe python=3.9 soxspipe -c conda-forge
```

Use should be told that you are about to overwrite the env, select yes.

smarttgit, 2023-05-12:
Maybe not a condo or python version problem. There is no median_absolute_deviation in scipy.stats
```
(soxspipe) bash$/Users/sjs/Dropbox/data>ls /Users/sjs/opt/anaconda3/envs/soxspipe/lib/python3.8/site-packages/scipy/stats
__init__.py                          _fit.py                              _rcont/                              _warnings_errors.py
__pycache__/                         _generate_pyx.py                     _relative_risk.py                    biasedurn.py
_axis_nan_policy.py                  _hypotests.py                        _resampling.py                       contingency.py
_biasedurn.cpython-38-darwin.so*     _kde.py                              _result_classes.py                   distributions.py
_biasedurn.pxd                       _ksstats.py                          _rvs_sampling.py                     kde.py
_binned_statistic.py                 _levy_stable/                        _sobol.cpython-38-darwin.so*         morestats.py
_binomtest.py                        _mannwhitneyu.py                     _sobol.pyi                           mstats.py
_boost/                              _morestats.py                        _sobol_direction_numbers.npz         mstats_basic.py
_common.py                           _mstats_basic.py                     _statlib.cpython-38-darwin.so*       mstats_extras.py
_constants.py                        _mstats_extras.py                    _stats.cpython-38-darwin.so*         mvn.py
_continuous_distns.py                _multivariate.py                     _stats.pxd                           qmc.py
_covariance.py                       _mvn.cpython-38-darwin.so*           _stats_mstats_common.py              sampling.py
_crosstab.py                         _odds_ratio.py                       _stats_py.py                         statlib.py
_discrete_distns.py                  _page_trend_test.py                  _stats_pythran.cpython-38-darwin.so* stats.py
_distn_infrastructure.py             _qmc.py                              _tukeylambda_stats.py                tests/
_distr_params.py                     _qmc_cy.cpython-38-darwin.so*        _unuran/
_entropy.py                          _qmc_cy.pyi                          _variation.py
```

smarttgit, 2023-05-12:
conda stubbornly won't update. 
But the reinstall worked 
```
(soxspipe) bash$/Users/sjs/Dropbox/data>soxspipe -v
v0.8.0
```

thespacedoctor, 2023-05-12:
I dug around and it may be that you installed conda with a version of python that is no longer supported and hence has no upgrade path. Python 3.7 support ends next month, so try this might work:

```bash
conda install -n base conda python=3.8
```

smarttgit, 2023-05-12:
The re-install upgraded to python 3.9.16 in the soxspipe environment : 
(soxspipe) bash$/Users/sjs/Dropbox/data/soxs/soxspipe-quickstart-demo/product/soxs-stare>python -V
Python 3.9.16

And all worked from there,  v.0.8.0 was installed and `_reduce_all.sh ` completed with all files as expected in the directories

smarttgit, 2023-05-12:
General comments : 
1. Seemed somewhat slow., but of course had to do all steps. When I deleted one of the VIS frames and ran it, it took 1m 10sec for the frame. Presume the others only need done once per night or batch ? Then 1m per frame is OK, but depends what is required per frame. SOXS should also be quicker than XShooter. 
2. Dead pixels (low response) in the mflat for NIR resulted in hot pixels in the science frame (division of low number) rather than the flat correcting them. Detector issue, SOXS NIR may be different but worth checking what XShooter data manual says about them   
3. Generally the sky subtractions look OK, although would need to see the extractions to tell. But top right of the VIS arm looked like it would have strong residuals. That part of the chip looked quite different to the others.

smarttgit, 2023-05-12:
That as on MacBook Pro, Apple M1, 16GB

thespacedoctor, 2023-05-12:
> 1. Seemed somewhat slow., but of course had to do all steps. When I deleted one of the VIS frames and ran it, it took 1m 10sec for the frame. Presume the others only need done once per night or [...]

Is this for the sky-subtraction? If so, the majority of the time is spent writing out the QC plots. I am going to turn off all plots by default and user can request them with an optional flag.

I am sure there is still quite a bit we can do to optimise for speed.

smarttgit, 2023-05-12:
Yes, the 1m10 sec is for skysub stage. Remove a sky sub frame : 
`(soxspipe) bash$/Users/sjs/Dropbox/data/soxs/soxspipe-quickstart-demo>rm ./product/soxs-stare/2019.08.31T00.29.36.007_VIS_1X1_SLOW_STARE_SAX_J1808.43658_SKYSUB.fits`
re-run : 
`(soxspipe) bash$/Users/sjs/Dropbox/data/soxs/soxspipe-quickstart-demo>time sh _reduce_all.sh `
..... models sky, subtracts and creates the sky subtracted frame again ....
`real	1m10.056s`

smarttgit, 2023-05-12:
Yes, the PDFs aren't necessary in operational mode. The skysub_quicklook.pdf for VIS and UVB have scale issue - no contrast. NIR ones are ok though.

smarttgit, 2023-05-12:
Think that's all I can do for now ? Looks good. Runs reliably, picks the files, applies the corrections as expected. Gives info on the steps as it goes. No user interaction, as expected for this step of processing.

thespacedoctor, 2023-05-13:
Thanks Stephen, useful feedback.

> Dead pixels (low response) in the mflat for NIR resulted in hot pixels in the science frame (division of low number) rather than the flat correcting them. Detector issue, SOXS NIR may be different but worth checking what XShooter data manual says about them

Do you mean the scattergun effect?

[image]

These are all in the XShooter bad-pixel mask. The XShooter NIR is riddled with these dead pixels and ignored in all processing steps. Eventually, I will set the pixel values to 0 or 1 (the bad-pixel map is found in extension 3 of all data frames).

closed 2023-08-09 by thespacedoctor