thespacedoctor/soxspipe #191: Upgrade to v0.8.0 does not work

jbodenst, 2023-06-25:
Hi all,

I'm trying to upgrade my installation from v0.7.2 using the explanation on the Quickstart webpage (https://soxspipe.readthedocs.io/en/master/quickstart.html).

However, running the command given there, it tells me 
_# All requested packages already installed._                                                                                               

while _soxspipe -v_ says: v0.7.2 ... 
Could you please let me know how I can install the newest version of the pipeline?

thespacedoctor, 2023-06-26:
Hi Julia,

Could you please open a new terminal window and then try this command:

> conda create -n soxspipe python=3.9 soxspipe -c conda-forge

Select 'yes' when asking to overwrite the current environment and then:

> conda activate soxspipe

If this works I'll update the quick start to use this as the default install.

jbodenst, 2023-06-26:
It works, thank you :)

closed 2023-08-09 by thespacedoctor