thespacedoctor/soxspipe #199: Tune the SOXS NIR recipe parameters to SOXS data (currently tuned to XSH)

thespacedoctor, 2023-06-30:


thespacedoctor, 2023-07-03:
I have iterated through various settings for the SOXS `disp-sol` recipe when processing real SOXS data. Here are the results:

| order deg  | wavelength deg | clipping sigma | clipping limit | mean res | res stdev |
| :------------ | :----------- | :----------- | :----------- |:----------- |:----------- |
| 2     | 2  | 5 | 10 | 24.95 | 16.17 |
| 3     | 2  | 5 | 10 | 10.62 | 9.51 |
| 3     | 3  | 5 | 10 | 1.31 | 0.94 |
| 4     | 3  | 5 | 10 | 0.81 | 0.94 |
| 4     | 4  | 5 | 10 | 0.33 | 0.24 |
| 5     | 4  | 5 | 10 | >1000 | >1000 |
| 4     | 5  | 5 | 10 | 0.26 | 0.22 |
| 4     | 5  | 7 | 10 | **0.25** | **0.20** | 

This is the best fit I could find:

```text
soxs-disp-solution:
    pixel-window-size: 10
    order-deg: 4
    wavelength-deg: 5
    poly-clipping-iteration-limit: 10
    poly-fitting-residual-clipping-sigma: 7
```

[image]

thespacedoctor, 2023-07-04:
Here is the order-centre tracing tuned as best it can be. As the background is ~1000 times the expected value, the trace to the right of the frame is destroyed.

[image]

thespacedoctor, 2023-07-04:
Order-edge fitting from MFLAT recipe. 

[image]

thespacedoctor, 2023-07-04:
The spatial solution is now tuned for SOXS data. Here are the results from various iterations on the polynomial degrees.

| order deg  | wavelength deg | wavelength deg | clipping sigma | clipping limit | mean res | res stdev |
| :------------ | :----------- | :----------- | :----------- |:----------- |:----------- |:----------- |
| 3     | 3  | 3 | 5 | 10 | 1.20 | 0.87 |
| 4     | 3  | 3 | 5 | 10 | 0.32 | 0.21 |
| 5     | 3  | 3 | 5 | 10 | 0.53 | 0.45 |
| 4     | 4  | 3 | 5 | 10 | 0.32 | 0.22 |
| 4     | 5  | 3 | 5 | 10 | 0.21 | 0.14 |
| 3     | 5  | 3 | 5 | 10 | 0.37 | 0.27 |
| 4     | 5  | 2 | 5 | 10 | **0.23** | **0.15** |
| 4     | 5  | 4 | 5 | 10 | 0.32 | 0.23 |

The chosen parameters (below and bold result above) are selected as they gave a good solution and the fit converges faster than the best parameters in the table above.

```yaml
soxs-spatial-solution:
    pixel-window-size: 10
    order-deg: 4
    wavelength-deg: 5
    slit-deg: 2
    poly-clipping-iteration-limit: 10
    poly-fitting-residual-clipping-sigma: 5
```

[image]

marcolandoni, 2023-07-04:
Super !!!
Seems to be very close to what obtained with Xshooter, right  ?

thespacedoctor, 2023-07-04:
The proof is always in the sky subtraction. The residuals are simply an indication of how well the polynomial fits the detection line positions ... it's the detected positions we can work on to improve.

closed 2023-07-05 by thespacedoctor