thespacedoctor/soxspipe #381: Wavelength stability in VIS

smarttgit, 2025-08-25:
From Paolo and Matteo's analysis 
We still have as a pending issue the stability of the wavelength in SOXS data. 

In short:
- we took the spectra of the same STD star (LTT 3864 aka CD-34 6792) observed over 5 nights (June 4, June 5, June 6, June 9, June 10)
- we reduced with soxspipe all the spectra with the same calibration dataset
- we computed the centroids of Hbeta and Halpha
- we found that the *relative* centroid shifts for Hbeta are up to 5.6 AA, while for Halpha are up to 2.0 AA
- there is an *absolute* centroid shift too -> Hbeta between 4.0 and 9.6 AA (247 - 592 km/s), Halpha between 5.0 and 7.0 AA (229 - 320 km/s)

while the absolute shift can be due to an intrinsic systemic velocity of the STD star (so I suggest to ignore it for the moment), the relative shift 
is rather big and we have to take care of it. Given our dispersion scale (of the order of 0.2 - 0.6 AA/pixel) I think we should have as a goal a stability << 1 AA. 

The observed instability in the wavelength calibration can be likely due to two factors:
- flexures effect
- pipeline (e.g. the ThAr atlas line, or the centroid algorithm)

Attached, some plots showinng the different line centroids over the observing epochs and how the centroids move w.r.t. telescope altitude, rotator angle (no clear correlations). 

A question: when deriving the dispersion solution does the pipeline use the sky lines too? Is it possible to have as a pipeline output the 1D sky spectrum? 

We think that a double check with the skylines can be useful for computing the dispersion solution. 

[image]

[image]

[image]

[image]

[image]

[image]

thespacedoctor, 2025-08-25:
> we reduced with soxspipe all the spectra with the same calibration dataset

Does this mean you used the same multi-pinhole frame (and therefore the same dispersion solution) for all the plots?

ferromatteo, 2025-08-26:
No, if I remember correctly we used the calibration files of the night itself for every night.

thespacedoctor, 2025-09-08:
The relative dispersion solution is now much improved. Here is a new plot of the H-beta line for all LTT 3864 data from May to June.

[image]

I still see a systematic redshift from the rest wavelengths across all pseudo orders, but chatting with Paolo, this is a genuine shift as the STD is travelling.

We will repeat the test now with an AGN, where we know the exact wavelengths (and redshift) of emission lines.

thespacedoctor, 2025-09-08:
From Paolo:

As I was discussing with Dave a few minutes ago, an ideal target for a careful check of the wavelength solution
is the AGN ESO 137-34: https://simbad.cds.unistra.fr/simbad/sim-id?Ident=ESO+137-34

We took at least a spectrum every night during the June run. The continuum is faint, but the narrow amission lines are clearly detected.

As you can see from the SIMBAD web page, the emission line redshift is z = 0.00876 (2626.01 km/s).

So, here the expected emission lines (*rest-frame* wavelength, angstroms):
[OII] doublet -> lambda,lambda = 3727,3729
[OIII] doublet -> lambda,lambda = 4959,5007
Halpha -> lambda = 6563
Hbeta -> lambda = 4861
Hgamma -> lambda = 4340
Hdelta -> lambda = 4102
Hepsilon -> lambda = 3970
Hzeta -> lambda = 3889
Heta -> lambda = 3835

the *observed* wavelength have to be the above values multiplied by (1+z) = 1.00876

csogeza, 2026-01-17:
Hi Dave, thank you for forwarding it to Paolo et al! 
Along these lines, Markus suggested to check if there are any alignments between the wavelength shifts and earthquakes in Chile, to see if that is the mechanical cause. I took a look at the earthquake history around La Silla / La Serena, and marked them on the plot, see below:

[image]

The text on the top of the plots shows the strength on the Richter scale and the approximate epicenter distance from the observatory.
It is not a 1-to-1 correspondence, but there might be some events that caused issues, like the one on the 19th of Dec. Of course, it is hard to assess this from the data online, since I don't know how these earthquakes felt on La Silla, but I tried to put all the ones within a couple of hundreds of km-s on the plot.

I will be on La Silla for the next two weeks, so will be happy to discuss this further there! Also let me know if you have specific ideas on what else to test!

csogeza, 2026-01-18:
Quick update: this seems to have something to do with the temperature of the CCD, or that of the common path (the added black curve marks the temperature values that were logged under 'HIERARCH ESO INS TEMP301 VAL' in the 'SSOL_MULTPIN_30.0S_SOXS_IMAGE.fits' headers, following the chat with Mirko):

[image]

The correlation plots between the wavelength differences and the CCD temperature

[image]

The correlation plots between the wavelength differences and the common path temperature

[image]

I have currently no clear idea of what causes the bimodality above a T difference of 1 C. My guess for now is that it has to do with how I calculate the wavelength differences per band per night (for now I am taking only a median across the bands)

thespacedoctor, 2026-10-21:
With daily arcs, we achieve a UV-VIS wavelength solution accuracy of **±0.5Å** (for the bluest regions, with greater accuracy achieved as wavelength increases)
With attached arcs, we achieve a UV-VIS wavelength solution accuracy of **±0.4Å** (for the bluest regions, with greater accuracy achieved as wavelength increases)

[Analysis here](https://github.com/thespacedoctor/soxspipe-wavelength-calibration-analysis/tree/develop/soxs_uvvis_analysis).

closed 2026-10-21 by thespacedoctor