thespacedoctor/soxspipe #394: UVBVIS extraction quality problems

amodigli, 2025-09-10:
I reduced all the available data with the pipeline version updated yesterday around 13:30 Munich time.
I still see a number of spectra where extraction quality need to be improved. I will attach some examples.
Please investigate source of problem, and unless it is to data of bad quality (in which case those should be flagged/hidden as bad in the archive), please improve reduction.

obj_108_Vir_SOXS.2025-05-15T03:11:19.072.fits_VIS.png: jump in continuum level still present
obj_AT2024aigi_SOXS.2025-09-09T09:28:21.397.fits_VIS.png: signal changing in range 10^-4-10^4 at super high frequency. Is that real? Several similar examples are for SAXSJ1808 object

obj_ESO_137-34_SOXS.2025-06-04T07:47:23.291.fits_VIS.png: clear quality problems

obj_CD-30_17706_SOXS.2025-05-14T06:31:12.679.fits_NIR.png: clear quality problems

obj_CD-44_12736_SOXS.2025-06-07T08:30:09.179.fits_NIR.png: extraction similar to case obj_AT2024aigi above

obj_STD_FLUX_SOXS.2025-05-08T04:33:51.085.fits_NIR.png: clear quality problems

obj_STD_TELLURIC_SOXS.2025-09-05T03:59:35.003.fits_NIR.png: extraction similar to case obj_AT2024aigi above
obj_V_V822_Cen_SOXS.2025-05-14T05:24:04.135.fits_NIR.png: extraction similar to case obj_AT2024aigi above

[image]
[image]
[image]
[image]
[image]
[image]

amodigli, 2025-09-10:
I reduced data from:

raw/2025-05-02/  raw/2025-05-06/  raw/2025-05-10/  raw/2025-05-14/  raw/2025-05-31/  raw/2025-06-04/  raw/2025-06-08/  raw/2025-06-12/  raw/2025-08-31/  raw/2025-09-04/  raw/2025-09-08/
raw/2025-05-03/  raw/2025-05-07/  raw/2025-05-11/  raw/2025-05-15/  raw/2025-06-01/  raw/2025-06-05/  raw/2025-06-09/  raw/2025-06-13/  raw/2025-09-01/  raw/2025-09-05/
raw/2025-05-04/  raw/2025-05-08/  raw/2025-05-12/  raw/2025-05-29/  raw/2025-06-02/  raw/2025-06-06/  raw/2025-06-10/  raw/2025-06-14/  raw/2025-09-02/  raw/2025-09-06/
raw/2025-05-05/  raw/2025-05-09/  raw/2025-05-13/  raw/2025-05-30/  raw/2025-06-03/  raw/2025-06-07/  raw/2025-06-11/  raw/2025-06-15/  raw/2025-09-03/  raw/2025-09-07/

The pipeline used corresponding calibrations (I guess closest in time, that is usually the best choice)

I limited plots to 6 examples and listed a few more with similar results. I suggest start checking those. Once those are solved please test on a larger data set.

File Naming convention: 
file='obj_'+object.replace(",","_").replace(" ","_")+'_'+arcfile+'_'+arm+'.png'
I attach python script

[plot_single.py](https://github.com/user-attachments/files/22250087/plot_single.py)

I attach updated plots (linear Y scale). Indeed in a few cases the log scale is not optimal. Still I see a number of quality problems (spikes: bad pixels/CRH not properly handled, holes, jumps)

[image]
[image]
[image]
[image]
[image]
[image]

I also attach the updated python script:

[plot_single_normal.py](https://github.com/user-attachments/files/22250599/plot_single_normal.py)

David, I hope these answers clarify your points.

smarttgit, 2025-09-10:
Related to the jumps between the orders, this seems to be a bug introduced into the code recently. Talked to Dave today, just recording here and not opening different issue as it is most likely related to Andrea's jumps. 

With the latest pipeline version v0.13.6  on 2025 09 08 data,  the *EXTRATCED_MERGED.txt file  does not match the *_EXTRACTED_ORDERS.fits file using the one of the  extractedFlux* columns. 

https://docs.google.com/document/d/1CVI5l79uUIWGnYUTSLUmTzynzzt2eWQUJlx2sSJcPK8/edit?usp=sharing

This is likely a pixel size scaling factor bug when the _MERGED.txt file is written out from the _ORDERS.fits file